econpy/torque #26: Noob Needs Help - No Front End?

duelldesign, 2015-02-12:
Hopefully this is just a dumb noob mistake.

I've followed the tutorial and didn't think that I had run into any issues. 

I can't seem to get the front end site to come up? I go to the address of my vps where I've configured this, but I just get the default ubuntu apache page?

Thanks for any help!

Coppan, 2015-02-12:
/session.php ???

duelldesign, 2015-02-12:
No luck with that, I'll just post it right here: http://162.248.165.237/torque/session.php

Coppan, 2015-02-12:
Maybe torque isnt in the root of the webserver?

duelldesign, 2015-02-12:
This is how the directory is laid out:

var/www/ 2 folders here: html and torque

so sessions.php would be located here: var/www/torque/sessions.php

Coppan, 2015-02-12:
your site says:

By default, Ubuntu does not allow access through the web browser to any file apart of those located in /var/www, public_html directories (when enabled) and /usr/share (for web applications). If your site is using a web document root located elsewhere (such as in /srv) you may need to whitelist your document root directory in /etc/apache2/apache2.conf.
The default Ubuntu document root is /var/www/html. You can make your own virtual hosts under /var/www. This is different to previous releases which provides better security out of the box.

Coppan, 2015-02-12:
move it to the "html" folder??

duelldesign, 2015-02-12:
Awesome, that works now. 

Now I just have to figure out why there aren't any sessions available to load. I should have a few in there.

Thank you so much for your help!

Coppan, 2015-02-12:
No worries mate! i guess the post.php couldent be reached until now?

Coppan, 2015-02-12:
i mean upload_data.php :-)

duelldesign, 2015-02-12:
I think you're probably right!

I'll test and let you know!

duelldesign, 2015-06-18:
Revisiting this! 

I've since gone to create a new install on a VPS and I don't see where there is a page provided to display the info. Am I doing something wrong again?