econpy/torque #30: Last data as JSON

ldittmar81, 2015-08-20:
Hello,

first I want to thank for the great job here. :-)

I need a way to get the last datas as JSON for my home automation. I want to know, when my car needs fuel. Example: I go mornigs to work and I see that there is no enouth fuel in the tank, because my son was too lazy to refuel!!! :-( If my home automation system would know that my car has not enough gas in the tank , then they could tell me right after the wake , so I have enough time to refuel.

This wold be realy great :-)

deennoo, 2015-11-24:
i try to do the same with domoticz, what's yours ?

deennoo, 2015-12-14:
Please fond below a bash code to export last session fuel and km remaining to 2 virtual device on domoticz.

#!/bin/sh
result=`mysql -u UTILISATEUR -pMOTDEPASSE torque -N -e "SELECT kff126b FROM raw_logs ORDER BY time DESC LIMIT 1"`
echo $result
kmrestant=`mysql -u UTILISATEUR -pMOTDEPASSE torque -N -e "SELECT kff126a FROM raw_logs ORDER BY time DESC LIMIT 1"`
echo $kmrestant
curl -g  "http://192.168.0.21:8080/json.htm?type=command&param=udevice&idx=376&nvalue=0&svalue=$result"
curl -g  "http://192.168.0.21:8080/json.htm?type=command&param=udevice&idx=377&nvalue=0&svalue=$kmrestant"

Run it once a day and enjoy