econpy/torque #35: Unknown column 'kd' in 'field list'

ruudvddries, 2016-03-20:
Hi Matt, hi all,

I have succesfully installed the very handy Opn Torque Viewer. Problem is that I can not load a session. Upon loading a session, I get the error " Unknown column 'kd' in 'field list' ". Do understand the error, as there is no 'kd' column in my SQL database. I have tried solving the error by examining the code but this didn't  get me anywhere. Can someone please point me in the right directions or does  someone have a solution?

Many thanks in advance!

Kind regards,

Ruud van den Dries

surfrock66, 2016-03-20:
Take a look at this page on my fork: https://github.com/surfrock66/torque/blob/master/scripts/create_torque_keys_table.sql

kd is the speed from the ODB.  In eronpy's version of torque, that's a default sensor, and when you load a session it pulls that PID automatically.  My guess is, you're not logging that in your app, so when it goes to populate it, there's nothing there.  Even if you made the column in the table, there's no data and some of the math fails.

One of the revisions in my fork is to modify the default choices and the whole plotting engine...the default is no plot, and when you DO go to plot there's only PIDs with data showing (and you aren't limited to 2 PID's, you can do 1 to infinity)

ruudvddries, 2016-03-20:
Thanks!

I Will give that a try! I expected 'kd' to be "Speed (OBD)" but my torque pid for that is k0d (default, I didn't change a thing) and I Have selected that as output.

I'll keep you posted!

Thanks for the fast respons.

Regards,

Ruud

surfrock66, 2016-03-21:
It could be different adapters use different PID codes?  In my fork I intended it to be more dynamic; I split the DB into 2 tables (and am about to add a third).  There's the raw_logs table, and now a keys table which tracks the pid, the description, whether there's data in it to show, and the minimum/maximum...and adds an interface to edit those.  When a new PID is detected, a new column should appear in raw_logs and a new row should appear in keys.

FYI, the third table I'm working on is a session table...I've been running this since 2014 (way before I forked it) and since my table is into the millions of rows...querying sessions takes forever.  To populate the session list, searching a table with sessionID, profile, start time, end time, etc is much more efficient, then only plotting would need to be done through the big DB, so queries should go quicker.

ruudvddries, 2016-03-26:
Hi surfrock66,

thanks for your forked version. It is working!

I had a spot of trouble getting it going. It appears there is a ')' to much on line 34 of upload_data.php. At the end of the line it should be "....$key))) {..."

Regards and thanks!

Ruud

surfrock66, 2016-03-28:
Good catch...I'd fixed it in a rev since then.  I'm doing a ton of debugging on the uploads of tracks...it's kind of a mess.  Torque (the actual android app) uploads data with HUGE $_GET requests (like 15,000 characters) and apache defaults to returning 414 responses with anything over 8190...meaning tracks were failing to finish uploading, DDoS'ing myself.  Fixed that...hopefully from that I'll be able to gather a prototype session from my apache logs to perform some more precise upload tuning.