econpy/torque #36: Map issue

galuhboy123, 2016-05-29:
good evening, my name is galuh boy from indonesia

i really interest in this OBD 2 project using php, and i have try to study this program since last couple month . I try to make only the "map" and "select session" only that will show in my browser (not include graph, table, etc),
but for some reason, the map doesn't work and not display on my browser, even thou i already selected the session.

can someone maybe have been study this program, have a solution?
or maybe which steps 1 by 1 that you prefer me to do for my idea ? (only showing maps and session)
need help thx.
#sorry for bad english (student)

regards boy

surfrock66, 2016-05-29:
I'll take a look when I get a bit; it's a holiday this weekend here in America so I have some family around, but if I get some time I'll take a look.  It shouldn't be so tough, I can show you the parts to comment out, but it may be Monday before I get to it.

surfrock66, 2016-05-29:
The unnecessary requires, sessions.php 10-11

You should be able to get rid of ANY <script> blocks that have jquery flot in them.  sessions.php lines 203-269

All the variable select code, sessions.php lines 338-358

The chart block, sessions.php lines 359-377

The data block, sessions.php lines 378-428

The export block, sessions.php, lines 429-447

I THINK that should do it...I haven't tested it (I use my own fork of this repo) but that should give you what you want.

galuhboy123, 2016-05-30:
thx  @surfrock66 
i'll wait for your help.

maybe i'll try bit more until your help comming .hehehe

thx again @surfrock66