econpy/torque #37: url.php

galuhboy123, 2016-05-30:
hai , im galuh boy from indonesia,
i may have question about file "url.php"

first:
what i know : the file will make the map having some line that using to **draw** our track position.
And what i want to ask is on first line that have $_SERVER function, 
what that function is purposed ? why the program should need that ?

second : 
the function of "mysql_escape_string" , i'm looking on mr.google 
there is no such function like that, the function should be "mysql_real_escape_string"
so what was that "mysql_escape_string" suppose to mean ?

anyone please help ...
#sory for bad english (student)

regards, boy

surfrock66, 2016-05-30:
1) This defines the scope of the cookie used to track the session information across multiple pageloads.  Since info is passed back and forth between php scripts, persistent information is stored in a cookie.  The line there says don't let the cookie persist after this session (so, it's not sitting around on your computer), and the $_SERVER directive really only means the cookie applies to just this directory...not another web directory on that server, like if you're me and you have multiple torque instances up (a test and a live one).  See here: https://secure.php.net/manual/en/function.session-set-cookie-params.php https://secure.php.net/manual/en/reserved.variables.server.php

2) It used to exist, but is now depreciated.  https://secure.php.net/manual/en/function.mysql-escape-string.php

galuhboy123, 2016-05-31:
@surfrock66 thx again buddy 
for helping me :dango: