econpy/torque #39: Upgraded to php7

monolog99, 2016-08-16:
Hi,

I had my Torque viewer working quite nicely up until last night when I upgraded my Ubuntu server and PHP got upgraded to v7, and that's where it all went wrong!! Not being a PHP guru I've been trying to hack my round it to get it working again with limited success. 

One of the key errors I keep getting is in get_sessions.php where it now complains about 

$timezone = $_SESSION['time'];

where it throws an error "Notice: Undefined index: time in /var/www/torque/get_sessions.php on line 14".

All the "mysql_" commands don't seem to work so have changed them to "mysqli_" which appears to make those errors disappear. Any pointers ?

Cheers, Tim.

surfrock66, 2016-08-17:
I solved this some time ago, but I don't remember spefically what my fix was.  Basically, session variables were created and referenced multiple times, and I was able to bypass that.  I think it was between these 3 commits to my fork:

https://github.com/surfrock66/torque/commit/005ba45d0c8eecde14c0cfb40b4d26d0f93b583d
https://github.com/surfrock66/torque/commit/2cc000d4a906476a0fb52306d3ed64207092c068
https://github.com/surfrock66/torque/commit/a3366bfdd75f942ad1b0eae9cd4b430a7322cdb4

monolog99, 2016-08-20:
Many thanks for the pointers. I've managed to get rid of all the errors now, following your code. Sadly though, no sessions show up. The raw_log table is filling up nicely, but the sessions table stays empty no matter what I try and not sure what else to try now after spending a few hours on it scratching my head.  Oh well, it was a nice tool whilst it lasted unless you can suggest any things I can try to debug what is happening. I *think* it's not meeting the "if (sizeof($_GET) > 0)" in the upload_data.php file which looks to be where the session table is populated. That said, that also looks like the same place the raw_log file is populated, which does appear to work, so am confused !

traindriverrev, 2016-11-18:
I'm having no sessions showing up too.

surfrock66, 2016-11-19:
Can you confirm that you have data logged in your database?