econpy/torque #42: Question: Upload continues after ride ends

frank-feulner, 2017-02-06:
Hi,

I have implemented your code to push my car's obd status data to a webserver.
Works great so far. Thank you very much for providing this!

What I don't get is why the app keeps pushing requests to my server long after I finish the ride. (It keeps uploading for roughly 20 minutes)

Do you have an Idea what might cause that behavior? Do I record too much/often?

**My settings:**
About 30 pids
Upload to webserver: Yes
Record selected PIDs: Yes
Data logging interval: 1 second
Web logging interval: 30 seconds

adel-s, 2017-02-06:
Hello!
How do you decide that you 'finish the ride'? Your car stops? You turned torque off?

surfrock66, 2017-02-06:
I experienced something like this, here's the solution I had for my specific problem (it's a copy/paste from the readme in my fork):

Gotchas

If you log a ton of PID's (as I do for debugging) you may encounter an apache bug; Torque uploads data through a huge $_GET request. Apache, by default, allows $_GET requests up to 8190 characters. My sample data upload was 13,619 characters long...this led to some data uploads returning 414 errors instead of 200 responses, which resulted in the app trying over-and-over to re-upload the datapoint, essentially DDoS'ing myself. For this; you can edit your apache configuration to allow any value you like; this is, in general, not recommended and should be set to limit a value as close as possible to what your longest query would be. On my configuration, I edited /etc/apache2/sites-available/000-default.conf and added the following line:

LimitRequestLine 15000

surfrock66, 2017-02-06:
To that end, look in your apache access.log and look at a request.  It'll show you the response code, but also the url being accessed.  You can just copy/paste that url into the browser, and if you get an "OK" back it's a different issue.

frank-feulner, 2017-02-09:
Hi everyone,
thank you for the quick responses.

The apache access log is unsuspicious: All the posts receive an OK. The char limit should also not be a problem since the calls range way below the 8.190 characters.

[image]

I forward the posts after slight modification to a ms azure event hub. In it's activity log, you can see a slump at 9:10 (that's when I drive into the parking garage and lose LTE connection for a few minutes). The whole activity after that is what makes me wonder. There are days when the app keeps sending posts for as long as 30 minutes after the ride's end. 

I have a feeling that there are still quite a few datasets in the queue of the app when I try to exit it after the ride because it takes too long for the individual post to be sent.

Therfore, I am curious what options I have in the app to make the app post less datasets and to reduce their size. (Plus, I use a German translation of the app and it does not reveal too much about what the particular upload options are supposed to do :( )