econpy/torque #46: Web server showing blank page

17100113, 2017-04-14:
Hi, I just cloned your web folder into my git and deployed it onto a web server but whenever I try to access the session.php it just shows a blank page with nothing. No HTML elements, nothing. Its completely empty. Can you kindly have a look at it and tell me what Im doing wrong? Also I had to change the name of the file session.php to index.php so that it opens that as default page. If I keep it to session.php it doesnt even deploy and gives an error. Im deploying it to heroku. Here's the link to the git and the heroku web server
https://github.com/17100113/torque/
http://torque.herokuapp.com/

kamikaze3891, 2018-05-24:
Try to change the PHP Version.
When PHP V7 is active the server shows blank pages.
Try to set the PHP version to V5.

Kind regards
Kamikaze

adel-s, 2018-05-27:
BTW, I'm using PHP v7 for Torque - and it's working perfect.