econpy/torque #47: Database Parameters

rokrapoorv, 2017-06-23:
I need help to create a lookup table for parameters that are being stored in the Server.For eg,
 What does Kf***** etc parameters mean? Some paramters such as Speed GPS are annonated.

surfrock66, 2017-06-23:
Check your apache logs.  If you're on linux, it's pretty easy to grep the access log for a phrase like mph.  There will be an entry which lists all the descriptions and codes.  Here's an example from mine:

cat access.log | grep Torque | tail -n 1 | head -c 3000
***.***.***.** - - [22/Jun/2017:13:32:24 -0700] "GET /torquetest/upload_data.php?eml=surfrock66@surfrock66.com&v=8&session=1498163542656&id=8a10dcdc03a8b509d9ff1dc42b855a4e&time=1498163543528&userUnitff1005=\xc2\xb0&userUnitff1006=\xc2\xb0&userUnitff1001=mph&userUnitff1007=&userUnitff122e=s&userShortNameff122e=0-100kph&userFullNameff122e=0-100kph%20Time&userUnitff1278=s&userShortNameff1278=0-100mph&userFullNameff1278=0-100mph%20Time&userUnitff124f=s&userShortNameff124f=0-200kph&userFullNameff124f=0-200kph%20Time&userUnitff1277=s&userShortNameff1277=0-30mph&userFullNameff1277=0-30mph%20Time&userUnitff122d=s&userShortNameff122d=0-60mph&userFullNameff122d=0-60mph%20Time&userUnitff122f=s&userShortNameff122f=1%2F4m%20time&userFullNameff122f=1%2F4%20mile%20time&userUnitff1230=s&userShortNameff1230=1%2F8m%20time&userFullNameff1230=1%2F8%20mile%20time&userUnitff1264=s&userShortNameff1264=100-0mph&userFullNameff1264=100-0kph%20Time&userUnitff1280=s&userShortNameff1280=100-200kph&userFullNameff1280=100-200kph%20Time&userUnitff1260=s&userShortNameff1260=40-60mph&userFullNameff1260=40-60mph%20Time&userUnitff1265=s&userShortNameff1265=60-0mph&userFullNameff1265=60-0mph%20Time&userUnitff125e=s&userShortNameff125e=60-120mph&userFullNameff125e=60-120mph%20Time&userUnitff1276=s&userShortNameff1276=60-130mph&userFullNameff1276=60-130mph%20Time&userUnitff125f=s&userShortNameff125f=60-80mph&userFullNameff125f=60-80mph%20Time&userUnitff1261=s&userShortNameff1261=80-100mph&userFullNameff1261=80-100mph%20Time&userUnitff1275=s&userShortNameff1275=80-120kph&userFullNameff1275=80-120kph%20Time&userUnit47=%20&userShortName47=A%20THR2&userFullName47=Absolute%20Throttle%20Position%20B&userUnitff1223=g&userShortNameff1223=Accel&userFullNameff1223=Acceleration%20Sensor(Total)&userUnitff1220=g&userShortNameff1220=Accel(x)&userFullNameff1220=Acceleration%20Sensor(X%20axis)&userUnitff1221=g&userShortNameff1221=Accel(y)&userFullNameff1221=Acceleration%20Sensor(Y%20axis)&userUnitff1222=g&userShortNameff1222=Accel(z)&userFullNameff1222=Acceleration%20Sensor(Z%20axis)&userUnit49=%20&userShortName49=PedalD&userFullName49=Accelerator%20PedalPosition%20D&userUnit4a=%20&userShortName4a=PedalE&userFullName4a=Accelerator%20PedalPosition%20E&userUnit4b=%20&userShortName4b=PedalF&userFullName4b=Accelerator%20PedalPosition%20F&userUnit62=%20&userShortName62=ActTorq&userFullName62=Actual%20engine%20%20%20torque&userUnitff124d=:1&userShortNameff124d=AFR(c)&userFullNameff124d=Air%20Fuel%20Ratio(Commanded)&userUnitff1249=:1&userShortNameff1249=AFR(m)&userFullNameff1249=Air%20Fuel%20Ratio(Measured)&userUnit46=\xc2\xb0F&userShortName46=Air%20temp&userFullName46=Ambient%20air%20temp&userUnitff1263=mph&userShortNameff1263=TripMSpeed&userFullNameff1263=Average%20trip%20speed(whilst%20moving%20only)&userUnitff1272=mph&userShortNameff1272=Trip%20Speed&userFullNameff1272=Average%20trip%20speed(whilst%20stopped%20or%20moving)&userUnitff1270=mb&userShortNameff1270=Baro&userFullNameff1270=Barometer%20(on%20

surfrock66, 2017-06-23:
CRAP sorry I just realized this is econpy's; I have a fork which has a PID table specifically to address this.