econpy/torque #50: incompatible with sql_mode=only_full_group_by

johnnymorris, 2017-08-06:
Hey,

So I have this working fine however when running in sql_mode=only_full_group_by (as default on MySQL version 5.7.17) an error is returned of: Expression #1 of ORDER BY clause is not in GROUP BY clause and contains nonaggregated column 'torque.raw_logs.time' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by

Anyone having the same issue and been able to resolve this?

Thanks.

kamikaze3891, 2018-05-23:
I have the same issue after change to a new hosting provider.
I´am working on a fix.
Can you please tell me on which system your server is running.
OS Ver.
PHP Ver.

Kind regards

kamikaze3891, 2018-05-24:
I have fixed the problem in my server.
I have deletet the entry only_full_group_by in the SQL server config file.
After delete this entry restart the sql server.
After the sql restart the website will work correctly.