econpy/torque #52: Sync problem after installing ssl on the server

kamikaze3891, 2018-05-18:
I must install an SSL certificate on behalf of the new data protection guidelines in germany.
After i have activated ssl on my webserver the torque app is no longer uploading data to the server.
is there any configuration that i have not seen?
Torque app is configurated and tells me all is ok when i start the sync test.
Also the web adress in torque has been updated to https

Kind regards
kamikaze

adel-s, 2018-05-20:
I faced this problem too. It is not about Torque viewer, it's related to TorquePro App. Application **can not** upload data via https. Even if you set upload url in app as https://yourserver/upload.php - it sends data to http://yourserver/upload.php url - check your web server logs.

Related topic at torque forum: https://torque-bhp.com/forums/?wpforumaction=viewtopic&t=8528.0 and https://torque-bhp.com/forums/?wpforumaction=viewtopic&t=7149.0

I solved this problem using my webserver (nginx) config - I opened http only for one file - upload.php, and It works.

kamikaze3891, 2018-05-20:
Thanks for this information.
I will check my configuration on my webserver and will check if it is possible to do the same with my webserver.
I must look if my plesk panel support this feature.

kamikaze3891, 2018-05-23:
After i have disabled SSL for the upload url the upload is working.

closed 2018-05-23 by kamikaze3891