econpy/torque #55: error in configuring websever

chunkhang123, 2020-11-17:
during the step of moving contents of  'web' folder to my webserver, the ubuntu 18.04 lts responded with 'could not stat web:no such file or directory.'
[image]
can anyone lend me some help, I'm new to this.

surfrock66, 2020-11-17:
Can you do "ls"?  I think the web directory isn't where you think it is.

chunkhang123, 2020-11-18:
thx for the reply, but im sorry that, what do you mean by "ls"? can I know more detail about that?
Is this what u mean?
[image]

By the way, I'm using Ubuntu 18.04 LTS on Windows 10 via WSL 2.

surfrock66, 2020-11-18:
well, there's your issue...the web directory isn't there, so it can't copy it to /var/www/torque.

Can you do 'ls -1 /var/www' and 'ls -1 /var/www/torque'?

chunkhang123, 2020-11-18:
> well, there's your issue...the web directory isn't there, so it can't copy it to /var/www/torque.
> 
> Can you do 'ls -1 /var/www' and 'ls -1 /var/www/torque'?

here is what I got back from the ubuntu
[image]

chunkhang123, 2020-11-20:
> well, there's your issue...the web directory isn't there, so it can't copy it to /var/www/torque.
> 
> Can you do 'ls -1 /var/www' and 'ls -1 /var/www/torque'?

Is the document 'web' needed to be downloaded or created from elsewhere?  since it's not spotted in the /var/www/torque.
If yes, can I know how? :-(

surfrock66, 2020-11-20:
First, can you confirm, is your web root (so, http://127.0.0.1/ or https://127.0.0.1/ ) /var/www or /var/www/html, I know different apache configurations use different web roots.

Your /torque/ directory looks just like mine.  When I go to https://server.domain.tld/torque/ I get the index.html in that folder.  The web directory is just to separate the scripts from the web root, you don't need a directory called "web" in your actual web root.

chunkhang123, 2020-11-21:
Does it mean my web root is in '/var/www/html' ? 
btw I cant access to http://127.0.0.1/ and https://127.0.0.1/, instead i went to http://localhost , hope they are the same.
[image]
[image]

surfrock66, 2020-11-23:
Sorry for the delay; yes it looks like the web root is /var/www/html instead of /var/www, so move the torque directory there

cd /var/www
mv torque html/

chunkhang123, 2020-11-24:
Hi, these are the latest update
[image]
[image]
After all that, I've edited the db_user, db_pass and db_port.
However, can I know in the next step, to enter the URL into the upload_data.php, which line to enter the URL into? cuz I couldn't find it just by looking for the obvious wording or signs.
[image]
Or is it in the other PHP file? I've went thru several of the PHP files as well, hopefully I didn't miss it.

surfrock66, 2020-11-24:
Before worrying about the data upload, can you check that going to http://localhost/torque/ works?

chunkhang123, 2020-11-25:
Thanks for the reply Surfrock 66
It looks like something is missing in my session.php file?
[image]

surfrock66, 2020-11-25:
Nope, that indicates there's an issue with the sql setup.  It's been a while since I set this up (my install has been working for like 6 years so the setup is a bit distant in my mind).  Can you confirm the database has been created and the tables are there?

chunkhang123, 2020-11-26:
This is the databases, but i didnt see any kind of tables in here.
Is the table supposed to be here or inside one of those databases?
can I know what type of the file is? like .csv or something else, this will ease my finding.
[image]
[image]
[image]

chunkhang123, 2020-11-26:
I guess you were saying this 
seems like this is the problem u mentioned
Can I know what's causing this?
[image]

surfrock66, 2020-11-27:
That's the one; can you actually enter the directory with the "create_torque..." file and run that same command, but "mysql -u khang -p < create_torque_log_table.sql", just make sure you can see the .sql file with ls before doing that.

chunkhang123, 2020-12-01:
hi suffrock66, sry for replying late, my laptop's hard disk crashed and everything on my laptop was gone.
I have got everything back to the latest step in the past few days..
[image]
[image]
I'm still doubting on whether the user-stuff (root or khang) will affect anything or not, I just did everything in root user just like before, but there was no such error coming out.

chunkhang123, 2020-12-02:
> Sorry for the delay; yes it looks like the web root is /var/www/html instead of /var/www, so move the torque directory there
> 
> cd /var/www
> mv torque html/

btw after the past few days experience, I realized that actually my web was actually moved and renamed into /var/www as torque (I executed the command "mv web /var/www/torque" and I have forgotten it)
so in this case, i just executed the normal command instead of the modified command u asked me to.
However, my http://localhost/torque/ has gotten worse, it becomes URL not found.
Can I know what might be the cause of it?
[image]

surfrock66, 2020-12-02:
I'm having a bit of trouble understanding the apache setup, can you post the contents of /etc/apache2/apache2.conf, /etc/apache2/sites-available/000-default.conf and /etc/apache2/sites-available/default-ssl.conf

chunkhang123, 2020-12-03:
So sorry for the confusion, these are the contents of respective .conf files. I put them in a word file, hope this will ease your helping hand.
[apache2 conf.docx](https://github.com/econpy/torque/files/5633948/apache2.conf.docx)
[Doc3.docx](https://github.com/econpy/torque/files/5633950/Doc3.docx)

chunkhang123, 2020-12-07:
> Sorry for the delay; yes it looks like the web root is /var/www/html instead of /var/www, so move the torque directory there
> 
> cd /var/www
> mv torque html/

[image]

Hi Surfrock66! I've revised a couple of issues discussed in the forum which one of those mentioned about the apache web root issue (quite similar to mine) and I tried to execute these 2 commands and turned out my http://localhost/torque worked!
I did not execute those 2 command lines when I got my laptop back from died harddisk becuz I was not sure that those 2 command lines should be executed or not. So sorry if any confusion was made.

BTW can I know which file here should I click as to view my real-time data from Torque Pro? or is there something else I should do before that?
FYI I just bought Torque Pro software and I yet to connect it to my OBD device, cuz I wanted to make sure every setting was perfectly done before I carry out any data logging.

In my Torque Pro software, I've inserted the Webserver URL as http://example.com/upload_data.php (as what shown in the main page) but this is not working. Should the word "example.com" be replaced with my IP address or something else?

surfrock66, 2020-12-07:
Hey, been offline for a few days, glad the web server is working.  

As for what to put in Torque Pro, let me try to explain because I think you might be a bit confused; if this is stuff you know then my apologies, but the questions make it sound like you're still learning about networking and DNS (which is good, everyone starts somewher).  Is this running on a server in your house, or do you have a server hosted somewhere like linode (or any cloud host) or something?  We need to point the app to your installation, but there is a lot of variation how that can be set up.  "example.com" is meant to just be an example for whatever domain name you use, but you could use the IP if you want.  This gets confusing though because if you're on normal residential internet...your IP changes all the time.  For example, I have Comcast in the US and my public IP to my house changes.  I have a dynamic dns service that runs so that, for me, "hda.surfrock66.com" always points to the IP of my house, there's a little program running on a server in my house checking if my IP changes, and if it does, it updates the IP associated with that domain.  There's LOTS of solutions to that problem, DynDNS is a popular one, noip is another, just research Dynamic DNS.  The alternative is to just put the IP of your house in torque and update it every time the IP changes, but most people don't do this because it's a lot of work and if your IP changes and you didn't notice, your server won't be reachable until you fix it.  An automated solution is usually preferred.

The other thing to remember is the IP inside your house is different than the IP outside of your house; inside the house uses one of the protected IP spaces ( https://www.arin.net/reference/research/statistics/address_filters/ ) and the one outside your house uses a public widely-accessible IP.  Because of this, when you're inside your network (on your wifi) you would connect directly to the IP of the server if it's hosted in your house (so, phone -> server over an IP like http://192.168.1.xxx) but OUTSIDE your house, you connection takes a different route (so, phone -> home internet provider like comcast -> modem -> server).  Because of this, you need to set up port forwarding...http traffic happens on port 80 and https happens on port 443, so you can usually set up something on your modem that says "Any requests that come to my house on port 80 need to go this this device on my network, any requests that come to my house on port 443 need to go to this device on my network".  It's called "Port Forwarding" but every modem is different so you'll need to look at the interface and the manual for the device in your house.  

Let me map this out one more time so it's hopefully easier to understand with some examples; the numbers here don't matter, they're just for demonstration:
Internal IP of the server in your house: 192.168.1.100
IP of your house on Monday: 123.124.125.126
IP of your house on Tuesday: 12.13.14.15
Dynamic DNS Example Domain: http://dyn.com/chunkhang123

On Monday, from your phone inside your house, if you open "http://192.168.1.100/torque/index.html" from your phone you'll get the torque web interface.  From your phone inside your house, if you open "http://dyn.com/chunkhang123/torque/index.html" it will not work because you don't access a server from inside your house using the address from outside your house (you can accomplish this with an internal DNS server but it's complicated and probably out of scope). If you do NOT have ports 80 and 443 forwarded in your modem to 192.168.1.100, and from outside the network you go to "http://dyn.com/chunkhang123/torque/index.html" on your phone you will get nothing.  If you DO have ports 80 and 443 forwarded in your modem to 192.168.1.100, from outside your house on your phone if you go to "http://dyn.com/chunkhang123/torque/index.html" you'll get the torque web interface.  If you DO have ports 80 and 443 forwarded in your modem to 192.168.1.100, from outside your house on your phone if you go to "http://123.124.125.126/torque/index.html" you'll get the torque web interface.  

On TUESDAY, from your phone inside your house, if you open "http://192.168.1.100/torque/index.html" from your phone you'll get the torque web interface.  If you DO have ports 80 and 443 forwarded in your modem to 192.168.1.100, from outside your house on your phone if you go to "http://dyn.com/chunkhang123/torque/index.html" you'll get the torque web interface (assuming your dynamic DNS is set up...it'll know the new public IP and point that domain name to your house).  If you DO have ports 80 and 443 forwarded in your modem to 192.168.1.100, from outside your house on your phone if you go to "http://123.124.125.126/torque/index.html" you'll get nothing.  If you DO have ports 80 and 443 forwarded in your modem to 192.168.1.100, from outside your house on your phone if you go to "http://12.13.14.15/torque/index.html" you'll get the torque web interface.  

Torque uploads data as it gets it, so as soon as the data is uploaded it's visible in the web app.  In reality, there's bit of a delay; I'd say under 5 minutes with a good phone connection.  I've been on a long 8+ hour drive and had my family use the torque app as a location tracker.  For live data, the torque app is good enough by itself; even when it's not uploading data, it's keeping an internal log and will show live data there.

chunkhang123, 2020-12-09:
Hi Surfrock66, thank you very much for the information, and these few days I've been researching on DDNS-related stuff, but I still face some issues here. I need ur help again to check is there any detail I've missed out or some basic mistakes I've made. :(
I'm using the DDNS provider NoIP
[image]
From this picture, I have doubts about the selection here, should I pick "DNS hostnames" or something else?

[image]
[image]
After that, I've followed the guidelines manual of setting up port forwarding for the router model in my house.

[image]
But the port 80 forwarding tester still showed no finding. Is there something I've missed out?

And the last question is, my ddns domain is http://khangtorque.ddns.net, if I'm gonna access the torque/index.html from outside of my house, is this the way I should type into the browser of my phone, http://khangtorque.net/torque/index.html ?

surfrock66, 2020-12-09:
Ok, to your last question, that is what I would expect, https://khangtorque.ddns.net/torque/index.html .

The DNS hostname thing should work, but I don't know why they have separate selections for 80 and Web; in theory web traffic is just ports, and ports can be anything, and dynamic dns is just the IP.  I'll over-explain it just for understanding...when you go to http://www.google.com you're really going to http://www.google.com:80 and when you go to https://www.google.com you're going to https://www.google.com:443.  You can do other things too, like for example, I use "airsonic" whcih has a web server on 4040, so I can go to https://hda.surfrock66.com:4040 to get to it.  The dynamic DNS name translates the name to the IP of the house, and you specify the port separately in the request.  The DNS name has nothing to do with the port and it should make all ports accessible by that IP.

It looks like the port forward is closed; this is tricky because every device is different.  For example, I have a comcast business class gateway, and these are the instructions for me: https://www.2mcctv.com/blog/2017_07_26-port-forwarding-smc-comcast-business-gateway-router/

If you know the model of your router/modem, you may want to google to see if other people have had issues with it.  I'm assuming the 192.168.8.105 is the LAN address of the web server in your house, and you've confirmed 14.1.224.210 is your public IP from your router.  You can go to https://www.whatismyip.com/ to confirm, or if you want to use the linux command line this command will do it: 

`wget http://ipinfo.io/ip -qO -`

chunkhang123, 2020-12-17:
Hi Surfrock66, sorry for the late reply. For the last whole week  I've been digging into port forwarding articles to configure the wifi router in my house, it is still not working :-( The port checking website still showing that port 80 is closed. Thus I am suspecting that my ISP is restricting the access on port 80 from the external network, maybe?

These few days I've tried software called "pagekite", not sure if you've heard of it, it provides service to expose my localhost to the internet, such that everyone can access to it. It sounds dangerous but I did give it a try, but all I get is it saying the localhost:80 is down.

Therefore I wonder here, is it possible for me to change the apache server to work on other ports, like port 8080, instead of on port 80? Then only I will use the other way to access my localhost from the devices outside my house.

BTW can I have your thoughts or opinion on the workability, if I let my laptop to connect to my mobile phone's hotspot, and launch apache web server (letting my laptop and mobile phone under the same network). Will this make everything easier, without the need of port forwarding?

  FYI my final aim of using Torque Pro is actually to show the REAL-TIME data (instead of logged record) on any web browser, this is part of the work in my final year report of my Bachelor Degree.

chunkhang123, 2020-12-19:
Hey Surfrock66! It's a good news this time!! 
I've tried with Ngrok instead of PageKite for my localhost tunnelling,  and I did it,  IT WORKS on my mobile phone outside of my house! 

But this lead me to another question, which php file do I have to insert the url for uploading data from Torque Pro? 
[image]

surfrock66, 2020-12-19:
Glad to hear it!  You need to go to session.php; I have all my shortcuts pointed directly at that.  Normally a webserver tries to send you to index.html or index.php when you go to the root of a folder, but since that's not there, it loads the directory (which is something you can disable but I don't know how off the top of my head).

chunkhang123, 2020-12-20:
hi Surfrock66, my session.php isn't showing anything .. and also when Torque Pro is uploading data, my server is showing 500internal server error. Can I know what might be the possible cause of it? will it be possibly caused by the misconfiguration in php? 
[image]
I have my data uploading url inserted in the Torque Pro apps as https://example.ngrok.io/torque/upload_data.php

chunkhang123, 2020-12-20:
hi Surfrock66, this morning I've searched thru the internet and found out there was some compatibility issues from php 7.x,  so I worked my way to downgrade my php7.2 to php 5.6
The session.php showed something this time, but not the content it intended tho. 

[image]

does this statement relate to some configuration inside mysql?

chunkhang123, 2020-12-21:
Hi Surfrock66! I've configured the sql_mode in my my.cnf file and it's finally working!! 
But can I know is there any approaches that we can get new data, automatically shown on the plot without needing to click refresh in the browser?

surfrock66, 2020-12-22:
Glad it's working, sorry for the delay, deep in work stuff before a holiday break.

Auto refresh isn't something there by default, but it'd be easy to add to session.php.  There's lots of ways to do it in lots of languages; here's some options:

https://stackoverflow.com/questions/11497611/php-auto-refreshing-page
https://stackoverflow.com/questions/11604261/refresh-page-automatically
https://www.internetearnings.com/auto-refresh-webpage-php-html-javascript/

chunkhang123, 2021-01-02:
Thank you very much Surfrock66 for ur precious time and all the information throughout the whole time!! Sincerely thank you from the bottom of my heart, now my system is working smoothly and perfectly! 
On the other side, let me make an apology for the late reply here since I was spending most of my time on enhancing the system.

And before I leave this beloved website, one last request from me, Mr Surfrock 66, can I include ur name in the acknowledgement session in my report as to appreciate all the helps I received from the professionals? 

Love from Malaysia, PEACE !

surfrock66, 2021-01-04:
No worries, feel free, hope things are going well out where you are!