econpy/torque #56: Torque API

chiavariniluca, 2022-07-27:
Has Torque any API's to query the db and extract the data?

surfrock66, 2022-07-27:
Not directly.  You can export to a csv or query the db directly in mysql, but there is no api for pulling data directly.

chiavariniluca, 2022-07-27:
Is there a guide/tutorial (or something like this) to do this?
Thank you in advance!

surfrock66, 2022-07-27:
There isn't a guide, but the actual data structures aren't that complex.  Take a look here, in this version of the app, the database is a flat structure:  https://github.com/econpy/torque/blob/master/scripts/create_torque_log_table.sql

I'll point you to this file in my fork, where I have more keys imported and have a keys table with descriptions.  My fork splits the raw data into multiple tables to help with giant data sizes, hence moving things to their own tables:
https://github.com/surfrock66/torque/blob/master/scripts/create_torque_keys_table.sql