econpy/torque #57: Export data by Torque

chiavariniluca, 2022-09-29:
I need to work with raw data and analyze it on my PC. Can I download, by Torque, data in one of the common formats (csv, json, xml, etc.)?

Thank you in advance!

chiavariniluca, 2022-09-29:
By the app or autmatically in another way?

surfrock66, 2022-09-29:
Both, the app has an export feature (though it's been years since I used it, you'd have to find that project for how) and the webapp has buttons:
[image]

chiavariniluca, 2022-09-29:
And can I extract this data in an automated way (i.e., without going through app or web app)?

surfrock66, 2022-09-29:
Not sure what you mean, this whole thing is a couple of components.  The torque app is it's own thing, and the author has his own web server solution and storage, I don't know of any automation within that.  This project is an API-compatible alternative web server for uploading the torque data to a place you own.  With that, you have access to the DB and could construct something to extract things on a schedule, or you could tweak the upload code to trigger the existing export routine when a session is complete.  Session completion is a bit tricky to detect though; it's a setting in the torque app when to generate a new session, but there's no "this session is over" event.  Basically the app says "we made a session, and are uploading data to it!  If the car stops generating data for more than x minutes, call it a new session!"  It could make sense to have some nightly php cronjob to say "look for sessions yesterday, then trigger the export function on those to a folder" but you'd have to figure it out, it's not part of this per-se, storing to the DB is where the automation ends.  Does that make sense?

chiavariniluca, 2022-09-29:
Yes, I think so. I saw your repo about Torque and I think that I need to implement my web server to upload my raw data on my database automatically.
Thank you for the help!